VincentWei/MiniGUI #6: use DT_CALCRECT in func DrawTextEx2, the text will be drawed

steprun, 2018-03-01:
I think DT_CALCRECT maybe only calculate the rect for the string. But in DrawTextEx2, the text will be drawed.

VincentWei, 2018-03-02:
Really? Please show me the code.

steprun, 2018-03-07:
if button dwStyle add BS_CENTER.
in paint_content_focus

    /*draw focus frame*/
    if (BUTTON_STATUS(pctrl) & BST_FOCUS)
    {
        focus_rc = *prc_cont;

        if (!BUTTON_IS_PUSHBTN(pctrl) && 
            !(pctrl->dwStyle & BS_PUSHLIKE) && 
            is_get_fmt)
        {
            dt_fmt |= DT_CALCRECT;
            DrawText (hdc, pctrl->spCaption, -1, &focus_rc, dt_fmt);
        }

        if (!is_get_fg)
            fg_color = GetWindowElementAttr((HWND)pctrl, WE_FGC_THREED_BODY);
        win_rdr->draw_focus_frame(hdc, &focus_rc, fg_color);
    }

Calculate text size format will add DT_CENTER

And in DrawTextEx2

If style have DT_CENTER, ctxt.only_extent = FALSE in while loop. And the text will be draw again.

VincentWei, 2018-08-31:
This is a bug and have fixed.

closed 2018-08-31 by VincentWei