VincentWei/MiniGUI #11: a suggest: support dynamic define variable mechanism for window instance

10km, 2018-09-01:
In my project, base `mgncs`, I design a dynamic define variable mechanism for a `mWidget `instance ,so called "Widget Local", 
see my blog for detail : https://blog.csdn.net/10km/article/details/81666413 .

I think it's a useful feature for general application design. Although  my code is base `mgncs`,but it's easy to modify for `minigui` , if you think necessary ,I can  contribute it for `minigui`. also if `minigui` have implemented alternative mechanism , please tell me.

VincentWei, 2018-09-03:
This is a good idea and good feature. Would you like to show your key API definitions?

VincentWei, 2018-09-03:
I want to introduce this feature for mGNCS.

10km, 2018-09-05:
the key API definitions is very simple, show in the article : https://blog.csdn.net/10km/article/details/81666413

all source is in the article .
two key function `fl_widget_local_set` and `fl_widget_local_get`

but,the source code style and naming rule is not fit for `mGNCS`,  need time to modified, and change to english comment.

VincentWei, 2018-09-05:
I have read your article. It is a good idea, but I prefer the following API definition:

`typedef (*CB_FREE_LOCAL_DATA) (DWORD local_data);
BOOL ncsSetWidgetLocalData (mWidget* owner, const char* data_name, DWORD local_data, CB_FREE_LOCAL_DATA cb_free);
BOOL ncsGetWidgetLocalData (mWidget* owner, const char* data_name, DWORD * data, CB_FREE_LOCAL_DATA* cb_free);`

The explanation:

1. The API should hide the implementation details, and keep it is simple and clear. So my version does not introduce the new data structure.
2. Use a const string as the name to the local data, so the caller does not need to call Str2Key before calling the functions.
3. Use the mGNCS style for API name.
4. Use DWORD data type to store the local data with a pointer wide. We should avoid to use the data type only work for 64-bit or 32-bit architecture,
5. Returns a BOOL for success or failure.
6. `ncsSetWidgetLocalData` will overwrite the old local data, and call the free callback function for the old local date if it was defined.
7. `ncsGetWidgetLocalData` will return FALSE if there is not a local data bound to the data name.

10km, 2018-09-05:
Fine!
I understand your explanation.
My implementation is unshaped, your verision is more compacted. 

about explanation 1 : hide new data structure is very necessary. caller needn't the local data structure

about explanation 2, limit only string as key, I feel needing more think: 
conside a scene, I design a class A derive from `mObject` as a widget local, the class A instance do something, needn't quoted by any object or function, we can call it as ‘anonymous variable‘，so it need not a string name as key. in my application, I add this type of object as widget local using the object pointer as key. I think need another function `ncsSetWidgetLocalDataEx` ,relplacing string key with `DWORD `key .

about explanation 4, I can explain : in my version,the reason of using  `int64_t` to store the local data is covering type `time_t` :), I agree your design.

10km, 2018-09-05:
about explanation  2,another solution: for `ncsSetWidgetLocalData`, when the input argument `data_name` is NULL,use the `local_data` as key....  :)

VincentWei, 2018-09-05:
> about explanation 4, I can explain : in my version, the reason of using int64_t to store the local data is covering type time_t :), I agree your design.

Since MiniGUI 3.2, DWORD is a 64-bit unsigned integer on 64-bit architecture, and a 32-bit unsigned integer on 32-bit architecture. We call this data type having pointer-wide, which is equal to the intptr_t date type introduced in C99. In deed, time_t is also such a data type, so it is enough to use DWORD for the local data. The developer can store a pointer, an integer, or a EPOCH time value in a local data slot.

> about explanation 2,another solution: for ncsSetWidgetLocalData, when the input argument data_name is NULL,use the local_data as key.... :)

The developer will be confused if we have different meaning for the arguments of a function. NULL for local_name should be considered as an invalid argument. Actually, I can not understand precisely the use scene you given. I think the developer should be responsible for the meaning of the name of a local data.

On the other hand, we should try to distinguish between facilities and mechanisms clearly. The API are facilities, the developer calls them and handle the mechanisms (how to use them) by themselves.

In addition, we need another function to delete the widget local data:

`BOOL ncsDeleteWidgetLocalData (mWidget* owner, const char* data_name);`

This function will remove the local data (and the CB_FREE_LOCAL_DATA callback will be called if it is not NULL).  As a special situation, If data_name is NULL, all local data bound the the owner will be deleted.

Thanks and regards!

closed 2019-01-28 by VincentWei