VincentWei/MiniGUI #19: Is Arabic supported in minigui?

imgtec, 2019-02-12:
Hola, Is these language supported in minigui?
1. Arabic
2. Thai

Cheers

VincentWei, 2019-02-13:
Quick answer:

1. Arabic: partially supported.
2. Thai: fully supported.

MiniGUI provides support for UNICODE, but it is not complete in MiniGUI 3.2. There is a major lack of edit boxes that support bidirectional (BIDI) layout. Except not supporting BIDI editing, you can use the MiniGUI API to handle and rendering Arabic text, as long as you have the right font.

In MiniGUI 3.4, which is actively developed, we will provide fully support of UNICODE 10.0. So almost all languages will be supported in MiniGUI 3.4.

We will officially release the first version of MiniGUI 3.4.0 about June of 2019. But now, you can visit the code at the branch of dev-3-4 in this repository.

Regards

closed 2019-02-13 by VincentWei

imgtec, 2019-02-17:
> In MiniGUI 3.4, which is actively developed, we will provide fully support of UNICODE 10.0. So almost all languages will be supported in MiniGUI 3.4.
is this saying that, Arabic will be supported in this release?

VincentWei, 2019-02-24:
> is this saying that, Arabic will be supported in this release?
Sure.

Actually, the support for Arabic (ISO8859-6)/Hebrew (ISO8859-8) is there since MiniGUI 3.0, and a single line editor for Arabic/Hebrew is also provided in MiniGUI 3.0.

The full support for Arabic needs the full implementation of UNICODE, this is the key enhancement of MiniGUI 3.4.