VincentWei/MiniGUI #22: 关于对minigui进行g2d加速的实现？

13824125580, 2019-09-22:
你好：
　　在一些基于低端处理器的应用场合，使用cpu做blit, scale或者二维blending会非常吃力，应用中一般会使用而二维加速单元GE（有些场合叫g2d)对gui进行加速，目前miniGUI对这部分操作都进行了软件实现，我的问题是，如果想用G2D硬件对minigui进行加速，应该修改哪些地方？miniGUI有统一定义加速接口 fops吗？还是要遍历整个miniGUI

VincentWei, 2019-09-26:
The interface for MiniGUI NEWGAL engine defines the operations to exploit the basic graphics hardware acceleration, such as video memory management, filling box, blitting with or without alpha.

Generally, you need to write a specific NEWGAL engine to implement the operations.

We implement a general NEWGAL engine for Linux DRI/DRM. You can refer to the code for an example. Some legacy engines, e.g., fbcon, also provide support for hardware acceleration.

13824125580, 2019-10-08:
Thanks for Your Information, this topic can be closed here!.
BTW, Is there any official support or issue discuss window for minigui Topic? it seems the github issuelist not fully used

VincentWei, 2019-10-08:
You are welcome.

FMSoft can provide the commercial support for the customer. However, I think that the issue list can be a good place to discuss some topics about MiniGUI. It may be a bit slow, but it is easy to use.

closed 2019-10-08 by VincentWei