VincentWei/MiniGUI #25: "__hb_extern_get_unicode_funcs"似乎找不到

meltsnow, 2019-11-06:
我将miniGUI4.0.2编译安装后，编译helloworld，提示：“undefined reference to `__hb_extern_get_unicode_funcs'”这个问题，即使我在gcc编译中添加-lharfbuzz，依然有这个问题
请问这是怎么回事

VincentWei, 2019-11-06:
You should use the modified harfbuzz library: https://github.com/VincentWei/harfbuzz

closed 2019-11-06 by VincentWei

khaledhosny, 2019-11-08:
Why do you need to patch HarfBuzz for this? there is already [`hb_font_set_funcs()`](https://harfbuzz.github.io/harfbuzz-hb-font.html#hb-font-set-funcs).

khaledhosny, 2019-11-08:
Sorry I meant [`hb_buffer_set_unicode_funcs()`](https://harfbuzz.github.io/harfbuzz-hb-buffer.html#hb-buffer-set-unicode-funcs)

VincentWei, 2019-11-11:
MiniGUI has own unicode functions. We need to patch HarfBuzz to use the Unicode functions provided by MiniGUI. Or, we have to link the functions provided by other library, for example, glib.

When we developing MiniGUI 4.0.0, HarfBuzz used a conditional compilation to determine which Unicode functions to use. We changed the code to use the functions implemented by MiniGUI.

I notice that in the new version, HarfBuzz has a new design to directly implement the Unicode functions in HarfBuzz itself. However, there are still duplicated implementations.

We may change this in the future according to the eventual implementation of HarfBuzz in order to avoid patching HarfBuzz.

Thanks!

khaledhosny, 2020-06-06:
You can build with `HB_NO_UNICODE_FUNCS` to avoid HarfBuzz compiling any of its Unicode functions, then `hb_buffer_set_unicode_funcs()` to set your own Unicode functions on the buffer.

VincentWei, 2020-06-08:
This is a really good solution.
Many thanks for your advice!