VincentWei/MiniGUI #29: 作为曾经老用户吐槽下, no zuo no die why you try

lanmanck, 2019-12-03:
minigui当年挺火的，还开源。后来大家给他踩坑差不多了后就闭源了，你说这狗血了吧，后来异军突起，minigui干不过别人又开源了，但是呢已经没有多少学生、开发者用了，现在居然出来刷存在感，真是no zuo no die a

VincentWei, 2019-12-09:
瞧，中国某些人就是这球德行，球嘴脸！

closed 2019-12-09 by VincentWei

VincentWei, 2019-12-09:
MiniGUI 发展至今，我们从所谓老用户获得的反馈屈指可数，也肯定不存在来自楼主的任何贡献，哪怕是一个缺陷修正或者拼写错误！看楼主的措辞，就是个典型的见不得别人好的小人，一个肯定混的不怎么样的不入流的工程师，一个丑陋的中国人！善良的人啊，没人会相信这样的人会做出任何的开源贡献。

MiniGUI 一直是开源存在的，中间停止更新，是因为那时的氛围下难以获得利润。2018年我们重装上阵，是因为有人愿意掏钱让我们继续发展，而不是为了刷存在感。

你以为我们是为了刷存在感搞开源？没钱我是不会搞的。这点上，我恐怕比楼主现实！

另外，有些当年用、现在用，还没怎么掏过银子的厂，现在是机会来还那些历史欠账了！希望楼主的公司主动联系我们缴纳历史欠账！

lanmanck, 2019-12-11:
拉倒吧你，我也就编译过一回玩了下，从来没有商用过。你好好学学人家rt-thread和thinkphp，别不作为就觉得别人建议刺眼。

VincentWei, 2019-12-11:
瞧瞧你的水平，开始标榜自己是老用户，还说帮我们踩坑了，最后自己打脸说只是编译过一回。

你这样的水平还来喷别人，中国人的脸都被你丢光了！

让我花功夫回复你，简直就是自取其辱！

赶快滚！