VincentWei/MiniGUI #33: 关于mGNCS、mGi的集成

cdslily, 2020-03-31:
您好，
我司使用的是rockchip的芯片，其中UI部分使用了miniGUI作为UI框架，但rk方面仅集成了miniGUI核心库，并没有集成mGi、mGNCS等附属库。咨询rk方面被告知其他组件需要我们自行移植。
在集成过程中提示
checking for MINIGUI... configure: error: Package requirements (minigui >= 3.0.10) were not met:
在网上我找不到合适的帮助信息。
是否可以得到您的帮助。

VincentWei, 2020-03-31:
I think that you have not correctly configure the cross-compilation command. Please refer to the following document:

https://github.com/VincentWei/minigui-docs/blob/master/user-manual/MiniGUIUserManualBuildingMiniGUI.md

I recommend that you first build MiniGUI on your Linux PC computer. Then cross-compile MiniGUI and the components for your target system.

I also suggest that you use MiniGUI 4.0/5.0 for new applications. We have prepared the complete building scripts for these versions:

https://github.com/VincentWei/build-minigui-4.0

or

https://github.com/VincentWei/build-minigui-5.0

Note that mGi is deprecated now. You can refer to mg-demos for a new input method sample:

https://github.com/VincentWei/mg-demos

cdslily, 2020-04-02:
好的，非常感谢！

closed 2020-04-02 by cdslily