VincentWei/MiniGUI #40: 5.0版本如果出现窗口套窗口的情况，显示就会出问题

meltsnow, 2020-04-22:
如题，在5.0版本下，proc模式中，如果有一个窗口套另一个窗口，小一点的窗口在大一点的窗口里面移动（不移出窗口外范围）就会出现如下图所示问题
[image]
而同样的代码和程序在4.0版本就不会出现
请问这是为什么，和新的合成图式功能有关吗？

VincentWei, 2020-04-28:
Thanks for this issue! The compositing schema is not enabled by default, so I think that you are not using the compositing schema. Is it right?

I guess this issue was caused by the change of the Z-Order operations. I will check this and fix it as soon as possible.

meltsnow, 2020-04-29:
> Thanks for this issue! The compositing schema is not enabled by default, so I think that you are not using the compositing schema. Is it right?
> I guess this issue was caused by the change of the Z-Order operations. I will check this and fix it as soon as possible.

对，我没有使用合成图式，4.0是没有这个问题的，5.0才出现，注意是大窗口上面有个小窗口，小窗口在没有移出大窗口范围时就会出现这种情况

VincentWei, 2020-05-12:
I have fixed this bug. Please check this in the branch rel-5-0.

Thanks for your bug report.

meltsnow, 2020-05-13:
谢谢，今天发现官网已更新5.0.1，经过测试该问题已解决

closed 2020-05-13 by meltsnow