VincentWei/MiniGUI #47: PROCS模式, 如果select 一直是 EINTR返回, 则MSG_TIME永远不会产生

steprun, 2020-09-03:
server_IdleHandler4Server中, 如果select 返回的 errno是 EINTR, __mg_check_expired_timers不会被调用, 那么就不会产生MSG_TIMER

VincentWei, 2020-09-03:
Thanks for your report.

Did you have any idea to fix this issue?

steprun, 2020-09-11:
--- a/src/server/server.c
+++ b/src/server/server.c
@@ -105,6 +105,10 @@ static void ParseEvent (PMSGQUEUE msg_que, int event)
     PKEYEVENT ke;
     MSG Msg;
 
+    __mg_check_expired_timers (msg_que,
+            SHAREDRES_TIMER_COUNTER - msg_que->old_tick_count);
+    msg_que->old_tick_count = SHAREDRES_TIMER_COUNTER;

VincentWei, 2020-09-14:
Thanks. Would you like to send a pull request?

VincentWei, 2020-09-21:
I have fixed this issue by calling `__mg_check_expired_timers` in `server_IdleHandler4Server`.
Please see the branch `rel-5-0`.

closed 2020-09-21 by VincentWei