VincentWei/MiniGUI #48: MoveWindow后 控件的surface没有更新到新为MainWindow surface

steprun, 2020-09-22:
__mg_update_dc_on_surface_changed 应该是在这里的更新为新的main win的surfce, 但是好像只有 privCDC 才会被更新

VincentWei, 2020-09-22:
Please provide sample code to reproduce the problem.

steprun, 2020-09-22:
代码不太方便发

steprun, 2020-09-22:
可以在 MSG_CREATE 中 对Mainwindow进行 MoveWindow , 然后在 获取 window 中通过 控件 的 GetWindowElementPixelEx, 由于 surface 已经被销毁, GAL_MapRGBA 会使程序崩溃

VincentWei, 2020-09-22:
Ok. Just show me the code to fix this issue.

steprun, 2020-10-10:
```
diff --git a/src/newgdi/gdi.c b/src/newgdi/gdi.c
index 52524a36..a222ef92 100644
--- a/src/newgdi/gdi.c
+++ b/src/newgdi/gdi.c
@@ -4413,6 +4413,7 @@ static void travel_children_with_priv_cdc (struct _travel_context* ctxt)
 
     child = (PCONTROL)(ctxt->parent->children);
     while (child) {
+        child->surf = ctxt->surface;
 
         if (child->privCDC) {
             ctxt->cb (ctxt, child);

```

VincentWei, 2020-10-12:
Thanks. I will merge the change.

closed 2020-10-12 by VincentWei