VincentWei/MiniGUI #51: V5 创建异型窗口不能生效, 只能控件生效, 3.0.12可以正常

steprun, 2020-10-12:
V5 创建异型窗口不能生效, 只能控件生效, 3.0.12可以正常
下面代码对控件使用是生效的, 但是对MainWindow使用不生效

```
    RECT rc;
    BLOCKHEAP cliprc_heap;
    CLIPRGN circle_rgn;

    GetWindowRect(hwnd, &rc);

    InitFreeClipRectList(&cliprc_heap, 50);

    InitClipRgn(&circle_rgn, &cliprc_heap);
    InitCircleRegion (&circle_rgn, 20, 20, 20);

    print_rgn(&circle_rgn);

    if (!SetWindowRegion(hwnd, &circle_rgn))
        printf("Error calling SetWindowRegion.\n");

    EmptyClipRgn (&circle_rgn);
    DestroyFreeClipRectList (&cliprc_heap);
```

steprun, 2020-10-12:
WS_EX_TROUNDCNS, WS_EX_BROUNDCNS 对窗口是有效的

VincentWei, 2020-10-12:
Ok. Got this bug report. I will check this issue and fix it.

Thanks!

VincentWei, 2020-10-12:
I have fixed this bug in rel-5-0 branch.

closed 2020-10-19 by VincentWei