VincentWei/MiniGUI #52: minigui not fount hb-ft.h and --with_ft2_includes is invalid

huohongpeng, 2020-10-16:
Hello, Mr Wei :
     miniGUI 5.0.3
     I have two issues.  
     1. hb-ft.h miss after make install harfbuzz, it's not be fount in include/harfbuzz.
     out info :  ....... shape-glyphs-complex.c:76:19: fatal error: hb-ft.h: No such file or directory
 #include <hb-ft.h>
     2.--with_ft2_includes is not effect. I guess is a bug  :
     file <configure.ac:2240> 2240     with_ft2_includes=""
                                                                                                                                                                Thanks

VincentWei, 2020-10-16:
1) Which version of harfbuzz are you using?

2) I will check this.

Thanks for your report.

steprun, 2020-10-17:
```
configure.ac
@@ -2236,7 +2253,7 @@ if test "x$build_ttf_support" = "xyes"; then
--
AC_DEFINE(_MGFONT_FT2, 1,
        [Define if support TrueType/OpenType fonts based on FreeType2])
 - with_ft2_includes=""
+ #with_ft2_includes=""
```

huohongpeng, 2020-10-19:
> 1. Which version of harfbuzz are you using?
> 2. I will check this.
> 
> Thanks for your report.
Mr Wei：
Thanks for your reply， Version is <harfbuzz-2.5.3>.
Thanks

VincentWei, 2020-10-19:
Thanks for your fix. I have merge it to `rel-5-0` branch.

For the `hb-ft.h` header file, I checked the source code of HarfBuzz 2.5.3, and found that there was the file in the source tree. I think that you did not installed hb correctly. Please check it.

huohongpeng, 2020-10-19:
Thanks Wei

closed 2020-10-19 by VincentWei