VincentWei/MiniGUI #56: 建议作者在minigui工程中加入input-event-codes.h

huohongpeng, 2020-11-09:
看到minigui中使用linux中的input-event-codes.h头文件，桌面的系统中会有这个头文件，但是当交叉编译时不会搜做桌面系统头文件路径，导致minigui编译出错。我看到libinput工程源码中就自己包含了这个文件，minigui是一个出色的可移植库，希望作者将input-event-codes.h包含在minigui工程中，提高可移植性。

ial/random.c:129:#include <linux/input-event-codes.h>
ial/linux-libinput.c:182:#include <linux/input-event-codes.h>

VincentWei, 2020-11-09:
Oh, no. This is a bad idea. As a middleware, we can not include such a system header in our source code tree.

closed 2020-11-09 by VincentWei

kmalloc8, 2021-01-05:
> 看到minigui中使用linux中的input-event-codes.h头文件，桌面的系统中会有这个头文件，但是当交叉编译时不会搜做桌面系统头文件路径，导致minigui编译出错。我看到libinput工程源码中就自己包含了这个文件，minigui是一个出色的可移植库，希望作者将input-event-codes.h包含在minigui工程中，提高可移植性。
> 
> ial/random.c:129:#include <linux/input-event-codes.h>
> ial/linux-libinput.c:182:#include <linux/input-event-codes.h>

最后这个怎么合进去？

VincentWei, 2021-01-05:
We have ignored this.