VincentWei/MiniGUI #62: WS_THICKFRAME 与 WS_TABSTOP 定义的值相同, 会导致 WS_TABSTOP 属性的控件有框

steprun, 2020-12-18:
/**
 * \def WS_THINFRAME
 * \brief Creates a main window with thin frame.
 */
#define WS_THINFRAME            0x00200000L

/**
 * \def WS_TABSTOP
 * \brief Indicates the user can set the input focus to the control by
 *        using Tab key.
 * \note This style is valid only for controls.
 */
#define WS_TABSTOP              0x00400000L

steprun, 2020-12-18:
应该是下面代码的问题, 没有判断是否有WS_BORDER属性
`
    else if(IsControl(hWnd))   // for control
    {
        if(win_info->dwStyle & WS_BORDER) {
            if(win_info->dwStyle & WS_THICKFRAME) {
                draw_3dbox(hdc, &rect, main_3d,
                       LFRDR_3DBOX_THICKFRAME | LFRDR_BTN_STATUS_PRESSED);
                for(i = 2; i < border + 2; i++) {
                    Rectangle(hdc, rect.left+i, rect.top+i,
                            rect.right-i-1, rect.bottom-i-1);
                }
            }
            else if(win_info->dwStyle & WS_THINFRAME) {
                draw_3dbox(hdc, &rect, main_3d,
                        LFRDR_BTN_STATUS_PRESSED);
            }
            else {
                draw_3dbox(hdc, &rect, main_3d,
                        LFRDR_3DBOX_THICKFRAME | LFRDR_BTN_STATUS_PRESSED);
            }
        }
        else if(win_info->dwStyle & WS_THICKFRAME) {
            for(i = 0; i < 2; i++) {
                Rectangle(hdc, rect.left+i, rect.top+i,
                        rect.right-i-1, rect.bottom-i-1);
            }
        }
        else {
            Rectangle(hdc, rect.left, rect.top,
                    rect.right-1, rect.bottom-1);
        }
    }
`

VincentWei, 2021-01-04:
Would you like to show the code correctly?

VincentWei, 2021-01-04:
I have changed the values of `WS_XXX` to avoid the possible conflict in the branch `rel-5-0`.
Please rebuild the code, and give me a response.

Thanks!