VincentWei/MiniGUI #66: 采用Minigui5.0，移植到嵌入式平台上，前台应用程序make时出现undefined reference to `hb_ft_face_create_cached'

forsafewu, 2021-02-02:
采用Minigui5.0，移植到nuc972平台上，前台应用程序make时，出现：
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libminigui_ths.so: undefined reference to `hb_ft_face_create_cached'
collect2: error: ld returned 1 exit status
Makefile:22: recipe for target 'all' failed
make: *** [all] Error 1

MAKEFILE文件内容如下：
CC := arm-linux-gcc-4.8.4
STRIP := arm-linux-strip
TARGET = ../build/ui
SRCS := main.c menu.c reset.c desktop.c dlg.c \
         menu_opt.c menu_view.c menu_disable.c menu_dev_status.c menu_program.c menu_set.c menu_net.c menu_debug.c \
         dlg_version.c program_dev_number.c program_dev_infom.c submenu_dev_infom.c data_struct.c program_link.c \
         program_dx.c program_detector.c program_pc.c program_send_loop.c real_event.c system_ipc.c setting_time.c \
         setting_printer.c setting_screensaver.c password.c dlg_memcpy.c submenu_concentration.c view_concentration.c \
         submenu_history.c view_history.c mask_addr.c mask_loop.c check_self.c prompt_box.c dialog_box.c opt_auto.c \
         check.c fire_confirm.c debug_password.c view_broken_line.c setting_relay.c startup_dev.c limit_mac.c limit_register.c \
         debug_usb.c net_info.c net_config.c opt_write_addr.c view_read_addr.c view_local_info.c view_loop_info.c print_manual.c \
         setting_lcd.c debug_fault.c debug_fire.c system_pyinput.c dlg_pyinput.c net_name.c setting_project_info.c dev_photo.c \
         dev_status.c debug_device.c mask_relieve.c mask_fault.c dlg_device_list.c 

LIBS= -I/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/include -Wall -Wstrict-prototypes -pipe -L/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib
PARM= -lpthread -lminigui_ths -lfreetype -lharfbuzz -ljpeg -lpng -lm -std=gnu99 -fgnu89-inline  
all:
	$(CC) $(LIBS) $(PARM)$(SRCS) -o $(TARGET)
	$(STRIP) $(TARGET) 
clean:
	rm -f *.o
	rm -f *.x
	rm -f *.flat
	rm -f *.map
	rm -f temp
	rm -f *.img
	rm -f $(TARGET)
	rm -f *.gdb

VincentWei, 2021-02-02:
Please check your installation of harfbuzz and the link options:

1. check whether there is the function exported in your harfbuzz library by using `nm` command.
2. try to adjust the order of libraries, such as: `-lpthread -lminigui_ths -lharfbuzz -lfreetype -ljpeg -lpng -lm`.
2. disable harfbuzz if you do not want to support complex script when you configure MiniGUI.

forsafewu, 2021-02-03:
配置MiniGUI时禁用harfbuzz，已解决上述问题。谢谢！

forsafewu, 2021-02-03:
配置MiniGUI时，禁用harfbuzz。已解决上述问题，谢谢！

closed 2021-10-19 by VincentWei