VincentWei/MiniGUI #68: Minigui5.0，用ministudio 写的程序,交叉编译不通过出现： undefined reference

forsafewu, 2021-02-21:
Minigui5.0，用ministudio 写的程序,交叉编译不通过，出现：
/usr/local/arm_linux_4.8/lib/gcc/arm-nuvoton-linux-uclibceabi/4.8.4/../../../../arm-nuvoton-linux-uclibceabi/bin/ld: warning: libminigui_procs-5.0.so.0, needed by /home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgncs.so, not found (try using -rpath or -rpath-link)
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgplus.so: undefined reference to `agg::font_engine_freetype_base::load_font(char const*, unsigned int, agg::glyph_rendering, char const*, long)'
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgplus.so: undefined reference to `agg::font_engine_freetype_base::prepare_glyph(unsigned int)'
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgplus.so: undefined reference to `agg::font_engine_freetype_base::flip_y(bool)'
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgplus.so: undefined reference to `agg::font_engine_freetype_base::width(double)'
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgncs.so: undefined reference to `mgTopmostLayer'
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgplus.so: undefined reference to `agg::font_engine_freetype_base::font_engine_freetype_base(bool, unsigned int)'
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgplus.so: undefined reference to `agg::font_engine_freetype_base::write_glyph_to(unsigned char*) const'
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgplus.so: undefined reference to `agg::font_engine_freetype_base::~font_engine_freetype_base()'
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgplus.so: undefined reference to `agg::font_engine_freetype_base::height(double)'
/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib/libmgncs.so: undefined reference to `Send2ActiveWindow'
collect2: error: ld returned 1 exit status
Makefile:21: recipe for target 'all' failed
make: *** [all] Error 1
root@ubuntu:/home/share/fs7510/app/ui/src# 

Makefile文件内容如下：
CC := arm-linux-gcc-4.8.4
STRIP := arm-linux-strip
TARGET = ../build/ui
SRCS := main.c menu.c reset.c desktop.c dlg.c \
         menu_opt.c menu_view.c menu_disable.c menu_dev_status.c menu_program.c menu_set.c menu_net.c menu_debug.c \
         dlg_version.c program_dev_number.c program_dev_infom.c submenu_dev_infom.c data_struct.c program_link.c \
         program_dx.c program_detector.c program_pc.c program_send_loop.c real_event.c system_ipc.c setting_time.c \
         setting_printer.c setting_screensaver.c password.c dlg_memcpy.c submenu_concentration.c view_concentration.c \
         submenu_history.c view_history.c mask_addr.c mask_loop.c check_self.c prompt_box.c dialog_box.c opt_auto.c \
         check.c fire_confirm.c debug_password.c view_broken_line.c setting_relay.c startup_dev.c limit_mac.c limit_register.c \
         debug_usb.c net_info.c net_config.c opt_write_addr.c view_read_addr.c view_local_info.c view_loop_info.c print_manual.c \
         setting_lcd.c debug_fault.c debug_fire.c system_pyinput.c dlg_pyinput.c net_name.c setting_project_info.c dev_photo.c \
         dev_status.c debug_device.c mask_relieve.c mask_fault.c dlg_device_list.c ncs_menu.c

LIBS= -I/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/include -Wall -Wstrict-prototypes -pipe -L/home/share/fs7510/bsp/minigui/libminigui-5.0.3/../build/lib 

PARM= -lmgncs -lmgutils -lmgplus -lstdc++ -lminigui_ths -lpthread -lfreetype -ljpeg -lpng -ldl -lz -lm -std=gnu99 -fgnu89-inline       
all:
	$(CC) $(LIBS) $(PARM)$(SRCS) -o $(TARGET)
	$(STRIP) $(TARGET) 

clean:
	rm -f *.o
	rm -f *.x
	rm -f *.flat
	rm -f *.map
	rm -f temp
	rm -f *.img
	rm -f $(TARGET)
	rm -f *.gdb

VincentWei, 2021-02-22:
The function `Send2ActiveWindow` is only available for MiniGUI-Processes runtime mode. Please check the runtime mode of your MiniGUI.

closed 2021-02-22 by VincentWei

forsafewu, 2021-02-24:
已解决，非常感谢！