VincentWei/MiniGUI #70: 初始化窗口时，控件多造成卡顿问题

kkdw789, 2021-03-09:
在嵌入式这类硬件性能比较差的情况下，初始化并显示窗口时，明显感受到控件加载时的缓慢和闪烁，用的是MGNCS控件集，不知是否有相应的解决方案或者代码案例供学习？

kkdw789, 2021-03-10:
现在的临时解决方案：
1、把所有控件都隐藏，也就是说进入窗口后不会感受到初始化控件的闪烁
2、然后把界面做成一张图片设置为窗口的背景图片
3、不再使用任何控件的嵌套和重叠
Q：以上方式虽然体验好了一些，但现在的的问题是切换界面时，依旧能明显的感受到一次全屏从上到下的刷新，体验还是不够好，不知道有没有什么办法来优化这一问题？

VincentWei, 2021-03-11:
Which NEWGAL engine are you using? If you use DRM (MiniGUI 5.0), you can enable double buffering to eliminate the flickers.

kkdw789, 2021-03-12:
> 您正在使用哪个NEWGAL引擎？如果使用DRM（MiniGUI 5.0），则可以启用双重缓冲以消除闪烁。
非常的感谢您百忙之中的回复，让我们在开发中找到了解决方向，我们采用的是MINIGUI 5.0，正在尝试使用DRM

但在配置过程中遇到一些问题，问题如下：
安装了libdrm-2.4.93进行交叉编译后，通过----enable-videodrm编译libminigui5.0，编译后出现：./configure: line 12176: syntax error near unexpected token `DRM,'
./configure: line 12176: ` PKG_CHECK_MODULES(DRM, libdrm >= 2.4, video_drm="yes",'
make: *** No rule to make target 'config.status', needed by 'stamp-h1'.  Stop.

Bob-hyb, 2021-03-12:
> > Which NEWGAL engine are you using? If you use DRM (MiniGUI 5.0), you can enable double buffering to eliminate the flickers.
> 
> 非常的感谢您百忙之中的回复，让我们在开发中找到了解决方向，我们采用的是MINIGUI 5.0，正在尝试使用DRM
> 但在配置过程中遇到一些问题，问题如下：
> 安装了libdrm-2.4.93进行交叉编译后，通过----enable-videodrm编译libminigui5.0，编译后出现：./configure: line 12176: syntax error near unexpected token `DRM,' ./configure: line 12176: ` PKG_CHECK_MODULES(DRM, libdrm >= 2.4, video_drm="yes",'
> make: *** No rule to make target 'config.status', needed by 'stamp-h1'. Stop.

你好,这个问题我刚好遇到过,这是由于缺少pkg-config导致的,安装pkg-config即可.

forsafewu, 2021-03-13:
root@ubuntu:/home/share/fs7510/bsp/minigui/libminigui-5.0.3# pkg-config --version
0.29.2
root@ubuntu:/home/share/fs7510/bsp/minigui/libminigui-5.0.3# 

已经安装pkg-config，还是不可以，仍然报错：
编译后出现：
 ./ configure：第12176行：意外令牌附近的语法错误DRM,'
 ./configure: line 12176: PKG_CHECK_MODULES（DRM，libdrm> = 2.4，video_drm =“ yes” ，'

kkdw789, 2021-03-16:
采用的是minigui5.0  也开启了drm，MINIGUI.cfg也进行了配置，代码里采用了双缓存来编写，其结果还是一样（在切换窗口时刷屏闪烁）
代码如下：
		case MSG_PAINT: {
			hdc = BeginPaint(hWnd);
			hdcMem = CreateCompatibleDC(hdc);
			FillBoxWithBitmap(hdcMem, 0, 0, 800, 444, &bmp_bkgnd_menu);
			sprintf(path_buf, "/app/picture/menu/%d.bmp", pos);
			PaintImageFromFile(hdcMem, photo_coor[pos].x,photo_coor[pos].y - Y_DEVIATION_TOP, path_buf);
			BitBlt (hdcMem, 0, 0, 800, 444, hdc, 0, 0, 0);
			DeleteCompatibleDC (hdcMem);
			EndPaint(hWnd, hdc);
			break;
		}

kkdw789, 2021-03-16:
对了  我们的GAL配置是  gal_engine=nuc970

VincentWei, 2021-03-18:
Who is the author of nuc920 engine?

VincentWei, 2021-03-20:
> 采用的是minigui5.0 也开启了drm，MINIGUI.cfg也进行了配置，代码里采用了双缓存来编写，其结果还是一样（在切换窗口时刷屏闪烁）
> 代码如下：
> case MSG_PAINT: {
> hdc = BeginPaint(hWnd);
> hdcMem = CreateCompatibleDC(hdc);
> FillBoxWithBitmap(hdcMem, 0, 0, 800, 444, [...]

There are two issues in your code:

1. Try to use `WS_EX_AUTOSECONDARYDC` style for double buffering of the main window instead of creating memdc.
2. Load the menu bitmaps ahead instead of calling `PaintImageFromFile`; this function is pretty slower than `FillBoxWithBitmap` or `BitBlt`.