VincentWei/MiniGUI #71: MiniGUI是否可以实现多个进程跑在同一个桌面上?

Bob-hyb, 2021-03-12:
目前我的理解是,每个 minigui app 都得链接 minigui lib ,也就是每个 app 都是单独具有一个背景桌面, 并没有办法让两个 app 显示在同一个桌面上,要实现这个肯定需要一个类似 gui server 的支持(如 xserver ), 但 minigui 仅仅只是一个库, 无法实现管理多个进程窗口的效果,不知道我的理解是否准确?

VincentWei, 2021-03-18:
It's not correct. miniGUI provides three runtime mode, under miniGUI-Procesdses mode, the mginit process acts as x-server.

Bob-hyb, 2021-03-25:
Let's say that there are two MiniGUI apps called A and B. They both link libminigui.so. Is it possible for them to appear as two windows on a same background?

VincentWei, 2021-03-26:
Yes. Under MiniGUI-Processes runtime mode, MiniGUI can act as a full-featured window system for desktop.

BTW, precisely, the client processes both link to libminigui_procs.so.

Bob-hyb, 2021-03-26:
Okay I see, but I have a question. How does the MiniGUI app know that there is a server already running?(for example app B mentioned above) Since a lib's data can't be shared across different apps. So I guess it requires some os support ?

VincentWei, 2021-03-27:
It is quite complex. IPCs, such as shared memory and semaphore, are required.

I recommend that you read some source code in MiniGUI, especially in the following directories:

- `src/server`
- `src/client`
- `src/kernel`

closed 2021-05-07 by Bob-hyb