VincentWei/MiniGUI #72: Windows builds?

IngwiePhoenix, 2021-04-21:
Hello there!

I want to try out MiniGUI and see it run on my local maschine and see what I can do with it. There are a few small ideas that I have but without a nice playground to test them, they will only stay ideas.

I noticed a `Makefile.msvc` there and, from reading the head of it, it's just an NMake file. Great! But it complains about `config.h` missing. And reading the CMake file reveals that MiniGUI wants PThreads.

So, is building on Windows no longer possible? Or will pthreads-w32 do? Or am I overlooking something?

Kind regards,
Ingwie

VincentWei, 2021-04-22:
Sorry, since 3.2, we no longer maintain the make files for Windows...

maikebing, 2021-05-07:
WSL 2  GUI已经发布， 有没有计划支持？

IngwiePhoenix, 2021-05-07:
I can not read chinese, so I will just make an assumption as to what you asked @maikebing 👍 

GUI apps in WSL2 can be achieved with either:
- X11, with the server running in Windows itself (VcXsrv, for instance)
- X11, with the server running in WSL and RDP or VNC being set up to connect to it

There is a new method being worked on as documented here: https://devblogs.microsoft.com/commandline/the-initial-preview-of-gui-app-support-is-now-available-for-the-windows-subsystem-for-linux-2/

VincentWei, 2021-05-08:
> WSL 2 GUI已经发布， 有没有计划支持？

You can try it. :)