VincentWei/MiniGUI #80: load_cursor_from_file: LoadCursorFromFile: bad file type: 512

maikebing, 2021-11-24:
oad_cursor_from_file: LoadCursorFromFile: bad file type: 512
load_cursor_from_file: LoadCursorFromFile: failed when loading cursor from /usr/local/share/minigui/res/cursor/d_arrow.cur
KERNEL>InitGUI: Count not init mouse cursor!
KERNEL>InitGUI: initialization failed; please check your MiniGUI configuration or resource.

maikebing, 2021-11-24:
高低位的问题。 
因为  512   的二进制是  0010 0000 0000

2  的二进制是       0010

maikebing, 2021-11-24:
在我修改 了 
大小端转换后，  

load_cursor_from_file: LoadCursorFromFile: bad cursor image width (-1) and height (-1)
load_cursor_from_file: LoadCursorFromFile: failed when loading cursor from /usr/local/share/minigui/res/cursor/d_arrow.cur

maikebing, 2021-11-24:
平台是   loongarch64

maikebing, 2021-11-24:
mips64 的机子就没这问题

VincentWei, 2021-11-25:
Please check the endianess.

maikebing, 2021-11-25:
重置了龙芯版的统信系统， 重新安装了所有环境， 编译并启动后提示  
```
GUI>LFRDR: failed to initialize default renderer's private info.
KERNEL>InitGUI: failed to init LF Manager!
KERNEL>InitGUI: initialization failed; please check your MiniGUI configuration or resource.
```

然后在 src/gui/lf_manager.c:864 设置断点， 
 if (rdr->init && 0 != rdr->init (rdr)) {
(gdb) p rdr->init
$1 = (int (*)(PWERENDERER)) 0xfff7c09c78 <init>
(gdb) p rdr->init(rdr)
$2 = -1
即  rdr->init(rdr) 返回-1 .

maikebing, 2021-11-25:
尝试了， 如果  禁用掉  cursor  不编译， 则 会出现上面的问题， 如果启用   -cursor  则 会 依旧出现   
```

load_cursor_from_file: LoadCursorFromFile: bad file type: 512
load_cursor_from_file: LoadCursorFromFile: failed when loading cursor from /usr/local/share/minigui/res/cursor/d_arrow.cur
KERNEL>InitGUI: Count not init mouse cursor!
KERNEL>InitGUI: initialization failed; please check your MiniGUI configuration or resource.

```

maikebing, 2021-11-25:
```
# lscpu | grep -i byte
Byte Order:          Little Endian

```

```
# dpkg-architecture | grep -i end
DEB_BUILD_ARCH_ENDIAN=little
DEB_HOST_ARCH_ENDIAN=little
DEB_TARGET_ARCH_ENDIAN=little

```

VincentWei, 2022-09-23:
应该是缺少对龙芯架构的处理导致的，在 `common.h` 中增加 `defined(__loongarch__)` 即可。

相关修改已添加到 `ref-5-0` 分支，可以试试 `rel-5-0` 上的代码。

maikebing, 2022-09-24:
好的， 我尝试一下。