VincentWei/MiniGUI #83: 使用5.0.9的minigui库时拖拽窗口会死锁右键菜单显示minigui消息对话框也会死锁

chenyinen, 2022-02-20:
之前用的3.2版本现在更新5.0版本跑mg-sample时发现以上问题，minigui运行在线程模式下！

(gdb) bt
#0  sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:85
#1  0x00007ffff7a41aaa in SendSyncMessage (hWnd=<optimized out>, msg=247,
    wParam=<optimized out>, lParam=140737488347888) at message.c:1754
#2  0x00007ffff7a1449b in MoveWindow (hWnd=0x60b9f0, x=<optimized out>,
    y=<optimized out>, w=800, h=<optimized out>, fPaint=<optimized out>) at window.c:5286
#3  0x00007ffff7a154bc in DefaultPostMsgHandler (lParam=2817498546992,
    wParam=240518168592, message=81, pWin=0x60b9f0) at window.c:2267
#4  PreDefMainWinProc (hWnd=0x60b9f0, message=81, wParam=240518168592,
    lParam=2817498546992) at window.c:2653
#5  0x00007ffff7a41308 in DispatchMessage (pMsg=pMsg@entry=0x7fffffffe3e0)
    at message.c:1386
#6  0x00000000004014bc in MiniGUIAppMain (argc=argc@entry=1,
    argv=argv@entry=0x7fffffffe588) at helloworld.c:209
#7  0x0000000000400d13 in main (args=1, argv=0x7fffffffe588) at helloworld.c:174
(gdb) info threads
  Id   Target Id         Frame
  3    Thread 0x7ffff5709700 (LWP 1729) "helloworld" 0x00007ffff76bb913 in select ()
    at ../sysdeps/unix/syscall-template.S:81
  2    Thread 0x7ffff5f0a700 (LWP 1728) "helloworld" pthread_rwlock_wrlock ()
    at ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_rwlock_wrlock.S:85
* 1    Thread 0x7ffff7fd8740 (LWP 1724) "helloworld" sem_wait ()
    at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:85

chenyinen, 2022-02-20:
我用valgrind跑了一下线程测试，发现z序这块有点异常
==49318==
==49318== Thread #1's call to pthread_rwlock_rdlock failed
==49318==    with error code 11 (EAGAIN: Try again)
==49318==    at 0x4C3510D: pthread_rwlock_rdlock_WRK (hg_intercepts.c:2355)
==49318==    by 0x4C379CE: pthread_rwlock_rdlock (hg_intercepts.c:2366)
==49318==    by 0x4EEC067: lock_zi_for_read (desktop.c:614)
==49318==    by 0x4EEC067: __mg_lock_recalc_gcrinfo (desktop.c:909)
==49318==    by 0x4F3E234: dc_InitDC (gdi.c:1969)
==49318==    by 0x4F3F240: GetDCEx (gdi.c:2343)
==49318==    by 0x4F3FC7E: GetDCInSecondarySurface (gdi.c:2876)
==49318==    by 0x4EC66CD: get_effective_dc (dc.h:687)
==49318==    by 0x4EC66CD: wndDrawNCButtonEx (window.c:914)
==49318==    by 0x4ECA3BC: DefaultNCMouseMsgHandler (window.c:1692)
==49318==    by 0x4ECA3BC: PreDefMainWinProc (window.c:2648)
==49318==    by 0x4EF5307: DispatchMessage (message.c:1386)
==49318==    by 0x40121B: MiniGUIAppMain (helloworld.c:187)
==49318==    by 0x400C32: main (helloworld.c:152)
==49318==
==49318== ----------------------------------------------------------------
==49318==
==49318== Thread #1 unlocked a not-locked lock at 0x658ACF8
==49318==    at 0x4C3559C: pthread_rwlock_unlock_WRK (hg_intercepts.c:2683)
==49318==    by 0x4C379EF: pthread_rwlock_unlock (hg_intercepts.c:2702)
==49318==    by 0x4F3E2BD: dc_InitDC (gdi.c:1995)
==49318==    by 0x4F3F240: GetDCEx (gdi.c:2343)
==49318==    by 0x4F3FC7E: GetDCInSecondarySurface (gdi.c:2876)
==49318==    by 0x4EC66CD: get_effective_dc (dc.h:687)
==49318==    by 0x4EC66CD: wndDrawNCButtonEx (window.c:914)
==49318==    by 0x4ECA3BC: DefaultNCMouseMsgHandler (window.c:1692)
==49318==    by 0x4ECA3BC: PreDefMainWinProc (window.c:2648)
==49318==    by 0x4EF5307: DispatchMessage (message.c:1386)
==49318==    by 0x40121B: MiniGUIAppMain (helloworld.c:187)
==49318==    by 0x400C32: main (helloworld.c:152)
==49318==  Lock at 0x658ACF8 was first observed
==49318==    at 0x4C34B79: pthread_rwlock_init_WRK (hg_intercepts.c:2119)
==49318==    by 0x4C379AD: pthread_rwlock_init (hg_intercepts.c:2134)
==49318==    by 0x4EF7F34: __kernel_alloc_z_order_info (zorder.c:247)
==49318==    by 0x4EF13A8: mg_InitDesktop (desktop-ths.c:115)
==49318==    by 0x4EE799A: InitGUI (init.c:550)
==49318==    by 0x400C1F: main (helloworld.c:152)
==49318==  Address 0x658acf8 is 120 bytes inside a block of size 35,096 alloc'd
==49318==    at 0x4C2DEC7: malloc (vg_replace_malloc.c:381)
==49318==    by 0x4EF7D78: __kernel_alloc_z_order_info (zorder.c:136)
==49318==    by 0x4EF13A8: mg_InitDesktop (desktop-ths.c:115)
==49318==    by 0x4EE799A: InitGUI (init.c:550)
==49318==    by 0x400C1F: main (helloworld.c:152)
==49318==  Block was alloc'd by thread #1

VincentWei, 2022-02-22:
Thanks for your detailed bug report! We will review and fix this bug asap.

chenyinen, 2022-02-25:
感觉这个问题有点严重，根据我的测试不仅仅在于上述场景，正常的movewindow都可能导致死锁的产生。

VincentWei, 2022-02-28:
You are right. We are fixing the bug.

WeSrc-Yin, 2022-03-08:
MiniGui是中国人做的开源软件吧。怎么全回复都整成英文的？

VincentWei, 2022-03-21:
We have fixed the bug: there is a deadlock introduced in v5.0.6. We have update the master branch and re-5-0 branch.

Thanks for your bug report and useful information!

closed 2022-03-21 by VincentWei