VincentWei/MiniGUI #91: 你好，关于一个编译的问题【急】

htk719809837, 2022-09-30:
前不久提出过一个问题， #88   现在我们是这样的，将minigui5.0.10下载到ubuntu，运行./autogen.sh以后，在移动位置放在我们自己的编译平台去编译（编译平台和svn为同一个目录），目前可以编译通过，但是这中间的拷贝minigui只能通过windows拷贝，如果通过linux拷贝会出现如下问题。

同样的，即使我们可以这样操作达到编译的目的，但是我们在本地定制好minigui5以后（只进行./autogen.sh，不编译），可以正常编译，那么我们将定制好不编译的minigui5上传至svn平台以后，在另一台电脑下载，也是报错如下。注意：如果同样在自己电脑上通过./autogen.sh并且定制好5.0以后，直接放在需要上传到svn的目录下编译是可以通过的！换而言之，本地定制好以后可以用windows压缩打包，发给另一台电脑，解压以后也可以编译。

现在的问题就是，svn应该是linux的文件系统，所以只要通过svn或者ubuntu之类的传输以后就不能编译。
是不是有解决方法，并没有编译过，而且只要不上传，都可以正常编译。目前这个问题很紧急，希望节前可以得到回复。

一下报错部分关键路径用xxx替代。

  * LibWebP includes:   
  * Pixman includes:    
  * FreeType2 includes: -I/xxxx/libminigui-5.0.10/./include/freetype2/
  * HarfBuzz includes:  
  * Dep. libraries:      -ljpeg -lpng16 -lz -ldl -lfreetype, -lm -lpthread

**********************start making minigui**********************
CDPATH="${ZSH_VERSION+.}:" && cd . && /xxxxx/libminigui-5.0.10/missing autoconf
 cd . && /xxxxxx/libminigui-5.0.10/missing automake-1.16 --foreign
/xxxxxx/libminigui-5.0.10/missing: 行 81: automake-1.16: 未找到命令
WARNING: 'automake-1.16' is missing on your system.
         You should only need it if you modified 'Makefile.am' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'automake' program is part of the GNU Automake package:
         <https://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <https://www.gnu.org/software/autoconf>
         <https://www.gnu.org/software/m4/>
         <https://www.perl.org/>
aclocal.m4:17: warning: this file was generated for autoconf 2.69.
You have another version of autoconf.  It may work, but is not guaranteed to.
If you have problems, you may need to regenerate the build system entirely.
To do so, use the procedure documented by the package, typically 'autoreconf'.
configure.ac:47: error: Autoconf version 2.65 or higher is required
aclocal.m4:703: AM_INIT_AUTOMAKE is expanded from...
configure.ac:47: the top level
autom4te: /usr/bin/m4 failed with exit status: 63
WARNING: 'autoconf' is probably too old.
         You should only need it if you modified 'configure.ac',
         or m4 files included by it.
         The 'autoconf' program is part of the GNU Autoconf package:
         <https://www.gnu.org/software/autoconf/>
         It also requires GNU m4 and Perl in order to run:
         <https://www.gnu.org/software/m4/>
         <https://www.perl.org/>
(CDPATH="${ZSH_VERSION+.}:" && cd . && /xxxx/libminigui-5.0.10/missing autoheader)
aclocal.m4:17: warning: this file was generated for autoconf 2.69.
You have another version of autoconf.  It may work, but is not guaranteed to.
If you have problems, you may need to regenerate the build system entirely.
To do so, use the procedure documented by the package, typically 'autoreconf'.
configure.ac:47: error: Autoconf version 2.65 or higher is required
aclocal.m4:703: AM_INIT_AUTOMAKE is expanded from...
configure.ac:47: the top level
autom4te: /usr/bin/m4 failed with exit status: 63
autoheader: '/usr/local/bin/autom4te' failed with exit status: 63
WARNING: 'autoheader' is probably too old.
         You should only need it if you modified 'acconfig.h' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'autoheader' program is part of the GNU Autoconf package:
         <https://www.gnu.org/software/autoconf/>
         It also requires GNU m4 and Perl in order to run:
         <https://www.gnu.org/software/m4/>
         <https://www.perl.org/>
make: *** [mgconfig.h.in] 错误 63
**********************end making minigui**********************

htk719809837, 2022-09-30:
我们这边的编译也是参考的上个问题中说到的编译步骤来的，只不过我们只用一个libminigui，其他组件并没有用到

VincentWei, 2022-09-30:
根据如下系统给出的如下提示做分析：

> WARNING: 'automake-1.16' is missing on your system.
>You should only need it if you modified 'Makefile.am' or
> 'configure.ac' or m4 files included by 'configure.ac'.

您使用的两个系统上的时间不匹配，导致在第二个系统中执行 `make` 命令时，系统发现 `confgure.ac` 文件被修改过，就会重新生成 `configure` 脚本。但第二个系统上的 `autoheader` 命令版本和你的第一个系统上的不一致，导致出现找不到`automake-1.16` 的问题。

解决方法：

1. 在第二个系统上执行 `./autogen.sh` 脚本，强制使用第二个系统中的 `automake` 命令重新生成 `configure` 脚本。
2. 确保两个系统上的 `autoconf` 以及 `automake` 版本一致。

htk719809837, 2022-10-12:
由于编译服务器问题，目前我们将源码打包以后上传svn，到另一台电脑解压出来即可，非常感谢！

closed 2022-10-12 by htk719809837