VincentWei/MiniGUI #96: 你好3.0升级到5.0以后有个接口变化想要咨询一下

htk719809837, 2022-10-27:
GetTabbedTextExtentPoint接口的变化是否影响了各个出参，应用中发现字符宽度的计算变化很大

VincentWei, 2022-10-28:
关注下逻辑字体风格以及相关接口的变化：

https://gitlab.fmsoft.cn/VincentWei/minigui-docs/-/blob/master/supplementary-docs/Using-Enhanced-Font-Interfaces.md

htk719809837, 2022-10-28:
还有一个问题，使用3.0的时候EDIT的插入符位置正常，但是使用5.0的时候，插入符光标会跑到当前字的中间，当然只有中文会出现，英文还是正常的，3.0就不会出现这个问题
[image]

VincentWei, 2022-10-31:
> esleft_input_char_refresh是不是这个函数修改引起的？

有可能，我们排查一下。

多谢反馈！

htk719809837, 2022-11-01:
请问插入符有定位到问题吗

htk719809837, 2022-11-03:
ttf
微软雅黑做过删减
GB2312

VincentWei, 2022-11-03:
我们使用 ttf 字体仍然无法重现。检查了 GetTabbedTextExtentPoint 函数的实现，和 3.0 相比未发现明显区别，只有几行多余的处理，移除了。请从 `rel-5-0` 分支更新后看看是否解决。

如果该问题仍然存在，可替换为未定制的字体文件，看是否是字体文件本身的缺陷造成的。还可以使用 UTF-8 编码，看看是否是因为字符编码而导致的。如果还存在，则需要提供可以复现该问题的代码以及字体，才能定位问题。

htk719809837, 2022-11-07:
你好，目前我在ubuntu上使用c:windows/fonts下面的微软雅黑字体，将其复制到ubuntu上，分别会有三个子字体，用这三个子字体进行测试，就会出现问题，同样的使用楷体就不会

[image]
[image]
[image]

目前代码是这样的
#include <stdio.h>

#include <minigui/common.h>
#include <minigui/minigui.h>
#include <minigui/gdi.h>
#include <minigui/window.h>
#include <minigui/control.h>
static PLOGFONT caption_font;

static LRESULT HelloWinProc(HWND hWnd, UINT message, WPARAM wParam, LPARAM lParam)
{
    switch (message) {
        case MSG_CREATE:
	hWnd = CreateWindowEx(CTRL_SLEDIT, 
		"微软雅黑",
		WS_VISIBLE,
		WS_EX_TRANSPARENT,
		401,
		48, 
		15,
		176,
		25,
		hWnd,0);
   	caption_font= CreateLogFont(NULL, "times", "GB2312", 
                    FONT_WEIGHT_BOOK , FONT_SLANT_ROMAN, FONT_FLIP_NIL,
                    FONT_OTHER_AUTOSCALE, FONT_UNDERLINE_NONE, FONT_STRUCKOUT_NONE, 
                    18, 0);
        SetWindowFont(hWnd, caption_font);
	SetWindowElementAttr(hWnd, WE_FGC_WINDOW, 0x00ff00);
            return 0;

        case MSG_DESTROY:
            DestroyAllControls (hWnd);
            return 0;

        case MSG_CLOSE:
            DestroyMainWindow (hWnd);
            PostQuitMessage (hWnd);
            return 0;
    }

    return DefaultMainWinProc(hWnd, message, wParam, lParam);
}

int MiniGUIMain (int argc, const char* argv[])
{
    MSG Msg;
    HWND hMainWnd;
    MAINWINCREATE CreateInfo;

#ifdef _MGRM_PROCESSES
    JoinLayer(NAME_DEF_LAYER , "mycontrol" , 0 , 0);
#endif

    CreateInfo.dwStyle = WS_VISIBLE | WS_BORDER | WS_CAPTION;
    CreateInfo.dwExStyle = WS_EX_NONE;
    CreateInfo.spCaption = "Hello, world";
    CreateInfo.hMenu = 0;
    CreateInfo.hCursor = GetSystemCursor(0);
    CreateInfo.hIcon = 0;
    CreateInfo.MainWindowProc = HelloWinProc;
    CreateInfo.lx = 0;
    CreateInfo.ty = 0;
    CreateInfo.rx = 480;
    CreateInfo.by = 480;
    CreateInfo.iBkColor = COLOR_lightwhite;
    CreateInfo.dwAddData = 0;
    CreateInfo.hHosting = HWND_DESKTOP;
    
    hMainWnd = CreateMainWindow (&CreateInfo);
    
    if (hMainWnd == HWND_INVALID)
        return -1;

    ShowWindow(hMainWnd, SW_SHOWNORMAL);

    while (GetMessage(&Msg, hMainWnd)) {
        TranslateMessage(&Msg);
        DispatchMessage(&Msg);
    }

    MainWindowThreadCleanup (hMainWnd);

    return 0;
}

#ifdef _MGRM_THREADS
#include <minigui/dti.c>
#endif

MINIGUI.cfg 对字体的配置如下：
[truetypefonts]
font_number=1
name0=ttf-times-rrncnn-*-64-ISO8859-1,ISO8859-2,ISO8859-5,ISO8859-7,ISO8859-8,ISO8859-9,UTF-8,GB2312-0,GBK
fontfile0=/usr/local/share/minigui/res/font/simkai.ttf
# The devfont now supports multiple font family names since V4.0.0.
name1=ttf-Source Sans Pro,SansSerif-rrncnn-0-0-ISO8859-1,UTF-8
# The path of font file can be relative to the current working directory.
fontfile1=font/SourceSansPro-Regular.ttf

htk719809837, 2022-11-07:
这个问题比较着急，大概什么时候可以修复好呢？

htk719809837, 2022-11-07:
另外的，对于阿拉伯语的BIDISLEDIT中文本的删除，现在会在插入符的地方多一个？

VincentWei, 2022-11-07:
> 这个问题比较着急，大概什么时候可以修复好呢？ 可以的话麻烦尽快回复一下，谢谢！

收到上述重新代码，排查和解决大概需要一周的时间。

> 另外的，对于阿拉伯语的BIDISLEDIT中文本的删除，现在会在插入符的地方多一个？

不清楚具体所指是什么问题。为提高沟通效率，所有问题的反馈，应该像上面一样给出明确的问题描述和重现代码。

VincentWei, 2022-11-07:
已修复。该问题在逻辑字体使用多个设备字体文件时会重现。

htk719809837, 2022-11-08:
多个设备文件是什么意思呢，我刚按照提交记录更新了一下发现好像还没有修复（网络问题，暂时克隆不了完整的代码），不知道是不是前几次提交仍有修改。不过可以确定，GetTabbedTextExtentPoint 我们这边用来计算字符串长度是错误的，正常长度为40的字符串，计算出来只有20多，和插入符可能同属于一个问题，希望可以再查一下看看。

在还有就是前面提到的。使用BIDISLEDIT，对阿拉伯语进行删除，会出现问号，这个字体同样使用上面的，demo我这边目前没时间调，希望在帮忙查一下。
[image]
[image]

htk719809837, 2022-11-08:
目前下了最新版本的代码，插入符问题已经修复

VincentWei, 2022-11-08:
好，bidiedit 和插入符透明的问题，我们这边排查一下。

htk719809837, 2022-11-08:
GetTabbedTextExtentPoint 我们这边用来计算字符串长度是错误， 这个问题有眉目了吗

VincentWei, 2022-11-08:
> GetTabbedTextExtentPoint 我们这边用来计算字符串长度是错误， 这个问题有眉目了吗

GetTabbedTextExtentPoint 这个函数和 GetTextExtentPoint 有相同的问题，其中 GetTextExtentPoint 的缺陷导致了编辑框插入符的问题。已经同步修复。如果这个函数还有问题，麻烦提供复现代码，否则很难排查。

另外，建议一个 Issue 中只提一个问题，还要描述清楚问题，最好给出复现代码。这样我们好排查、跟踪和管理，否则会降低我们的沟通效率。比如这个 Issue 当中，把 GetTabbedTextExtentPoint 的问题和插入符问题放在一起说，我们一直以为说的是同一个问题……

htk719809837, 2022-11-09:
好的，目前还剩两个问题

1、bidiedit，最好使用上面提到的微软雅黑测试
2、 插入符透明，我们这边根窗口用的是透明色	pWinInfo->iBkColor = PIXEL_transparent;

htk719809837, 2022-11-09:
貌似没有看到修复的提交.....

VincentWei, 2022-11-10:
> 这个提交好像提交到了master分支......

嗯 ，的确。已经同步到 rel-5-0 分支。

closed 2022-11-15 by htk719809837