VincentWei/MiniGUI #97: 关于minigui4.0 在ubuntu 22.4LTS上编译出现的问题

BowinChow, 2022-10-29:
您好，开发者，
  我在对cell-uxdemo在Ubuntu 22.40LTS(22.04 LTS)上编译出现下面错误，请问如何解决呢？（在Ubuntu18.04LTS编译可以通过）。
checking for minigui >= 3.2.0... no
configure: error: Package requirements (minigui >= 3.2.0) were not met:

No package 'minigui' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables MINIGUI_CFLAGS
and MINIGUI_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
make: *** 没有指明目标并且找不到 makefile。 停止。
make: *** 没有规则可制作目标“install”。 停止。

VincentWei, 2022-10-31:
请检查是否正确安装了 MiniGUI。另外，请使用 5.0 系列，4.0 系列是一个中间版本。

要在 Ubuntu 等 Linux 发行版上安装 MiniGUI 及其组件、示例程序等，请使用 build-minigui-5.0 仓库中的脚本：

https://github.com/VincentWei/build-minigui-5.0#readme

BowinChow, 2022-10-31:
魏博士您好，
  感谢您的回复！
祝好

closed 2022-10-31 by BowinChow