VincentWei/MiniGUI #98: 关于自定义组件toast相关问题

BowinChow, 2022-10-29:
您好开发者，
我在minigui中的activity里面想实现一个类似于android toast的组件。我的思路是：首先将文字/图片绘制组合成一个组件，当出发条件出现时就绘制这些piece,那么过程应该是：兴建一个container ->panelpiece->addContent(imagepiece/textpiece/gradientpiece)。当我想要延迟一段时间使这些组件消失的时候，我会新建一个线程，在此线程延迟一段时间，然后使用postmessage发送消息到主线程，主线程由ncsSetComponentHandler注册的函数执行delContent（targetPiece）的操作，然后使用Panel_invalidPiece()刷新ui。
但是奇怪的是，虽然delContent()的返回值为true, Panel_invalidPiece()并没有回调onPaint()方法，所以之前的组件并未消失。请问这样该如何处理呢？
附上伪代码：
static void onRefreshUI(paras){

panelpiece->delContent(imagePiece);

panel_invalidPieces(panelpiece);

 }
void postMyMessage(){
sleep(5);
postMessage(...,MSG,...);
}

//在主线程中（实际上能够回调到delContent,且返回值为true，但是界面上依然有图标，并且没有回调onpaint方法，如果同步执行
则可以删除。）
{
nscSetComponentHandler(component, MSG,onRefreshUI);

thread task(postMyMessage, para);
task.detach;

}

VincentWei, 2022-10-31:
请检查 `panel_invalidPieces` 函数的实现，是否最终调用了 `InvalidateRect` 函数将窗口客户区的指定矩形设置为无效（脏矩形）。只有包含有无效区域，才会产生重绘消息，并最终调用 `onPaint` 方法。

BowinChow, 2022-10-31:
魏博士您好，
  今天我尝试了您的方法，发现实际上在PanelPiece_invalidatePiece中确实调用了invalidateRect方法，但是奇怪的时依然没有去除图标。我把invalidateRect的第三个参数的FALSE设置为TRUE实现了图标的消失，目前原因未知。
很感谢您的回复。
祝好！

VincentWei, 2022-11-01:
`InvalidateRect` 函数的第三个参数含义是是否擦除背景，通常需要传入 `TRUE` 来擦除背景。目前看，这个 panel 的实现代码在没有内容的情况下，只是跳过了绘制，所以在传入 `FALSE` 的情况下，图标并不会被擦除。您的这个改动是正确的。

closed 2022-11-01 by VincentWei