VincentWei/MiniGUI #99: 你好，咨询一下关于多语言的问题

htk719809837, 2022-11-02:
目前我们就使用minigui5.0.10一个核心库，想做到中文切换其他语言不重启设备，有没有什么方案呢，目前我们切换语言需要设备重启

htk719809837, 2022-11-02:
包括切换阿拉伯语，自动翻转，有没有一整套的多语言方案呢

VincentWei, 2022-11-02:
这没啥特殊的呀，动态切换界面使用的文本字符串就可以了。可以使用 GNU 的 gettext 接口，选择装载不同的 mo 文件。

如果要同时支持阿拉伯等复杂语言，就要全面使用 MiniGUI 5.0 当中的 Unicode 相关接口，不能再使用以前针对中文和英文的文本处理和绘制接口了。

另外，设备字体和逻辑字体要做一些新的初始化。即使不重启设备，也要关闭当前已创建的窗口，然后重新初始化。

htk719809837, 2022-11-02:
有没有一个例子可以参考呢
---另外，设备字体和逻辑字体要做一些新的初始化。即使不重启设备，也要关闭当前已创建的窗口，然后重新初始化。
我们想做到和安卓一样，点击即可换，有具体的解决方案吗

htk719809837, 2022-11-02:
以及使用 Unicode 相关接口有没有什么参考资料呢

htk719809837, 2022-11-02:
动效实现（图片动态出现，和ppt的那种动画效果），5.0有什么实现建议？3.0可以通过双缓存+Bitblt实现，但是5.0不行

VincentWei, 2022-11-03:
> 有没有一个例子可以参考呢

可以看看 mg-demos 仓库中的 cbplusui，实现了语言的动态切换功能：

https://github.com/VincentWei/mg-demos/tree/master/cbplusui

> 以及使用 Unicode 相关接口有没有什么参考资料呢

可以看看 Unicode 官网，CSS Text Level 3 规范等：

http://www.unicode.org/
https://drafts.csswg.org/css-text-3/

> 动效实现（图片动态出现，和ppt的那种动画效果），5.0有什么实现建议？3.0可以通过双缓存+Bitblt实现，但是5.0不行

我记得您使用的多线程模式，这个模式下，原来在 3.0 上的所有功能，在 5.0 上并没有取消或者限制。5.0 版本上，我们提供了多进程模式下的合成图式（compositing schema），这个功能可以提供窗口级的切换特效。但这涉及到整个系统架构的重新调整（比如从多线程调整为多进程模式），另外所有窗口都会有自己独立的缓冲区，内存使用会加大，因此，这对您来讲会是个挑战。

> 3.0可以添加效果吗

没明白指的是什么效果。如果是界面的切换效果，您不是提到“3.0可以通过双缓存+Bitblt实现”吗？

VincentWei, 2022-11-03:
MiniGUI 5.0 合成图式的效果，可以看看下面的介绍文章以及其中的视频：

https://hybridos.fmsoft.cn/blog/hybridos-rk3308

LindaWenorTaolin, 2022-11-09:
想咨询一下：ANIMATION怎么显示背景透明的gif图？
当前窗口透明、设置ANIMATION控件为透明属性后，加载背景透明的gif图，显示出来控件背景是白色矩形

VincentWei, 2022-11-10:
应该是 ANIMATION 控件的背景色默认为白色导致，可以修改其背景色为透明色，也就是 Alpha 分量为 0 的颜色。

LindaWenorTaolin, 2022-11-10:
可以修改其背景色为透明色---------有什么具体的修改建议吗？
是不 setup_anim_mem_dc 接口需要做些改动？how？