VincentWei/MiniGUI #100: 关于创建透明主窗口（mainwindow）

BowinChow, 2022-11-03:
魏博士您好，
  请问如何将一个activity的mainwindow设置为透明？cellphone-ux-demo中有无参考案例呢？
祝好！

VincentWei, 2022-11-03:
主窗口的透明，有很多不同的需求，其实现要分别考量：

1. 在嵌入式系统中，如果要让主窗口透明，其目的是为了透出图形层下面的视频层，则一般通过将主窗口的背景色设置为硬件支持的透明色即可，比如 Alpha 通道为零。
2. 在一般场景下，如果想让主窗口变透明，且能看到该主窗口下面的其他主窗口中的内容，则有两种方法实现。第一种方法，利用 MiniGUI 的不规则窗口，这时候，使用 SetWindowRegion 或 SetWindowMask 函数设置窗口的区域或遮罩；这个方法支持所有的 MiniGUI 运行模式，但只能实现全透明。第二种方法，使用 MiniGUI 多进程模式下的合成图式（compositing schema），并通过设定主窗口的不同合成模式（透明、半透明等）来实现。第二种方式可以实现半透明等特殊效果，还可以通过定制合成器来实现窗口切换特效，但对硬件性能（尤其是图形性能）的要求较高。

具体示例：

1. mg-samples 仓库中 `src/nr_window.c`：使用 `SetWindowMask` 函数。
2. mg-tests 仓库中的 `api/setwindowregion.c`：使用 `SetWindowRegion` 函数。
3. mg-tests 仓库中的 `compositor/` 目录：使用合成图式并定制合成器。注意要使用合成图式，必须将 MiniGUI 配置为多进程模式，并开启 compositing 配置选项。

BowinChow, 2022-11-03:
魏博士您好！
  首先感谢您耐心的回复！
  由于刚接触minigui, 所以对其应用不是特别熟悉。请见谅。您说的第一点：对于下层视频层透明是采用哪一个api设置呢？
我看到了setbkcolor(),但是没有通道参数的设置。在下面的activity模板中，是否可以添加WS_EX_TRANSPARENT实现对于下层视频层的透明呢？
static NCS_MNWND_TEMPLATE mymain_tmpl = {
    NCSCTRL_MAINWND,                                // Window class name
    1,                                              // Window identifier
    ACTIVITY_X, ACTIVITY_Y, ACTIVITY_W, ACTIVITY_H, // Window location and size
    WS_NONE,                                        // Window Style
    WS_EX_NONE,                                     // Extension style of the window
    "demo",                                      // Window title
    NULL,                                           // Initialization property
    NULL,                                           // Renderer information
    mainwnd_handlers,                               // Window event handler
    _ctrl_tmpl,                                     // Template structure array pointer of the child window
    TABLESIZE(_ctrl_tmpl),                          // Number of the child window
    0, 0, 0                                         // Additional data the window, icon sentence handle and menu bar sentence handle
};

VincentWei, 2022-11-03:
创建主窗口时通过指定 iBkColor 或者绘制时，通过 SetBrushColor 指定画刷颜色，然后再调用 FillBox 函数。若要根据颜色分量获得对应的像素值，调用 RGBA2Pixel 函数。

另外，可参考 #92