VincentWei/MiniGUI #102: 关于透明窗口，panelpiece管理的控件删除后未消失的问题

BowinChow, 2022-11-08:
您好魏博士，
  我在设置窗口透明时发现如果该窗口的container设置透明，异步delContent()也成功了，并且InvalidateRect()也调用了onPaint方法，但是组件在界面上还是未消失。同https://github.com/VincentWei/MiniGUI/issues/98。
  但是如果我设置了一个background或窗口不透明，上述问题不会发生，控件可以异步删除。已经排查很久，并且感觉各项流程没有问题，我们组其他人确实也遇到过问题。
祝好！

BowinChow, 2022-11-09:
“这位仁兄，本仓库 Issue 不是用来提供技术支持的场所。”

--完全理解，诚恳地就我之前提issue的方式内容以及方式致歉。但是目前MiniGUI尚未一个供开发者自由讨论的平台（就我的认知来说）。

------------------------------------------------------

“要获得技术支持，请联系我公司销售部门（致信 [[sales@minigui.com](mailto:sales@minigui.com)](mailto:sales@minigui.com)），购买商业技术支持以获得相关资格。”
  --我个人十分尊重版权，但是就目前的研究来说，我不能够决定是否获取商业版权。

----------------------------------------------

“另外，根据您以往所提的问题推测，您目前的项目是基于 cell-phone-demo 的代码做的改动，有关这个演示程序的问题已经超过了技术支持的范畴，要知道这只是一个演示程序，我们不可能对所有的开源代码提供无限的保证和支持。因此，就目前您提到的这些问题，您需要自己排查处理。“

--结合自身的开发体验来说，很多基础功能可在demo中的实例得到启发，并且可以看到关于MiniGUI 3.0的pdf programming instructions/MiniGUI 5.0 docs。但是，由于项目是比较庞大的，很多要开发的新场景（e.g., toast）参考已有demo确实有些逻辑不通（就我个人而言无论是上述资源还是在google上很难得到一个查询的结果，这些api像是积木，但是如何使用是一种陌生的排列组合，对这种问题的引导或者提供一个规范我认为肯定对开发者很有帮助。例如android developers）。

关于排查问题，目前我能做的就是对比之前demo，对其流程插入日志，很多情况下得到的反馈是正常的，这不得不让我怀疑：这些api的使用是否遵循某种我不知道的规范。

-----------------------------------------

最后，欢迎在本 Issue 区提交 MiniGUI 及其组件相关的缺陷报告。谢谢！

--后续我会严格遵守这一规范，谢谢提醒。但是我想提出一个个人建议：能否建立一个关于minigui的官方的developer community （if already established, please kindly let me know.）。那么这样开发者可以在此平台上交流非bug性的问题。
-----------------------------------------

--写在最后：一个操作系统的源码确实复杂与需要耐心去维护，中间的工作量与所需的资源肯定也非常宏大，MiniGUI给了我们开发者一个十分珍贵的平台。也很感谢魏先生亲自回答我们的问题！上述有一些开发体验及建议应该能够体现部分开发者的心声。希望MiniGUI越来越好！

VincentWei, 2022-11-09:
购买商业技术支持，并不是要求同时购买商业许可，而且有些客户可能已经通过间接方式获得了商业许可，比如使用全志或者瑞芯微处理器的客户。

GitHub 平台的 Issue 区其实就是一个比较好的平台，我们针对其他客户的支持也是通过这个平台进行的。但对未购买技术支持服务的开发者，我们不会积极处理其问题。另外，就算有个独立的开发者社区，估计也不会有除了我们之外的人主动回答技术问题。毕竟大家都忙，人家有没有义务不是？

另外，您提问题方式存在一个问题：试图通过文字描述的方式将问题描述清楚，但这几乎不可能。遇到问题的最好办法，就是将相关代码写成一个非常简单的、只依赖 MiniGUI 及其组件的程序，让这个程序来复现你遇到的问题，这样理解问题本身就会很容易，同时也可以排除其他一些因素的干扰，排查起来也会容易一些。

但不管如何，既然您的项目比较大，建议还是联系我们的销售购买技术支持服务。

BowinChow, 2022-11-09:
"另外，就算有个独立的开发者社区，估计也不会有除了我们之外的人主动回答技术问题。毕竟大家都忙，人家又没有义务不是？"
--我想表达的意思是一个论坛，而不是一个强制性的回答社区，例如stackoverflow。开发者遇到的问题归结，汇总，开发者相互交流。例如我遇到的这个问题，也许其他使用MiniGUI的开发者会有对此问题的兴趣。（新加入的discussion模块是否可以讨论此问题呢？）

VincentWei, 2022-11-09:
> "另外，就算有个独立的开发者社区，估计也不会有除了我们之外的人主动回答技术问题。毕竟大家都忙，人家又没有义务不是？"
> 
> --我想表达的意思是一个论坛，而不是一个强制性的回答社区，例如stackoverflow。开发者遇到的问题归结，汇总，开发者相互交流。例如我遇到的这个问题，也许其他使用MiniGUI的开发者会有对此问题的兴趣。（新加入的discussion模块是否可以讨论此问题呢？）

嗯，您可以试试。