VincentWei/MiniGUI #105: 关于阿拉伯语遗留的问题

htk719809837, 2022-11-10:
你好，阿拉伯语删除的问题仍有部分为解决，并且新发现一个问题，希望可以这周内修复，我们这边比较着急

1、在edit控件为透明色的时候，输入任意东西，文字有默认的白色底色；

2、在输入后发送回车消息给当前edit，然后在输入一些字符，随后点击删除，发现会存在一两个字符删不掉；
这里的键盘我们是自己实现的，点击了对应的按键会往edit框里发送对应的按键消息。
[image]
[image]

VincentWei, 2022-11-10:
好，我们排查一下。不过这几天比较忙，只能周末处理了。

几个问题先确认一下：

1. “在edit控件为透明色的时候”：指的是编辑框的背景色 Alpha 为 0，可以透过屏幕看到视频层的内容吗？
2. “发送回车消息给当前edit”：这个编辑框是单行编辑框，应该过滤掉回车，您自己可以尝试修改下。另外，输入的字符中是否包含有 ISO8859-6 字符集的阿拉伯字母？如果有，还要提供输入的字符值（十六进制），否则我们这边没有阿拉伯输入法，无法重现。

htk719809837, 2022-11-10:
1、第一个不是的，可以看一下图一，我们是在static上叠加了背景图，包括放大镜，然后在画一个BIDISLEDIT上去。说白了图一能看到的都是用static放进去的背景图.只有输入的地方用的BIDISLEDIT。
这个edit的创建如下：是一个透明风格的
hWndCtrl = CreateWindowEx (CTRL_BIDISLEDIT,
			   "",
			   WS_CHILD | 自定义靠左属性 | ES_TIP | WS_VISIBLE,
			   WS_EX_TRANSPARENT, \
			   USER_SEARCH_CTRL_SEARCH_EDIT, 
			   X, 
			   Y, 
			   W, 
			   H, hWnd, (DWORD)0);

2、输入字符按理说都是acsall码，我们输入的都是一些字母和数字，目前没有做阿拉伯文字的输入支持，复用了英文的键盘。
确实用的是CTRL_BIDISLEDIT，单行的，上面的创建方法写的。这边过滤你们那加一下可以的，我们这边尽量不去改动代码以及5.0源码

VincentWei, 2022-11-15:
请从 `rel-5-0` 分支获取最新代码，并确认是否修复。

如果插入回车还有问题，需要告知回车字符是通过 MSG_KEYDOWN 还是 MSG_CHAR 发送的，包括 wParam 和 lParam 参数是什么。

htk719809837, 2022-11-15:
已修复

closed 2022-11-15 by htk719809837