VincentWei/MiniGUI #109: cross-compiling for mips

gowrirengu, 2022-12-14:
Hi
we are trying to cross compile minigui for MIPS architecture. So, I tried this: **./configure CC=mips-linux-gnu-gcc --prefix=/opt/mips-linux-gnu-ingenic-gcc7.2.0-glibc2.29-fp64/ --build=i386-linux --host=mips-linux-gnu --target=mips-linux-gnu** configure command in all folders with the configure file. In the cell-phone-ux-demo folder, this configure command is showing **"configure: error: SQLite3 is not installed! Run "sudo apt install libsqlite3-dev" to install it if you are using Ubuntu"**, even though Sqlite3 is installed in the system. And while trying make in other directories, where configure command ran without errors, it is showing like this

**Making all in bmp
make[1]: Entering directory "/minigui/build-minigui-5.0/minigui-res/bmp"
make[1]: Nothing to be done for "all".
make[1]: Leaving directory "/minigui/build-minigui-5.0/minigui-res/bmp"**

FYI, We are using Ubuntu 20.04.4 LTS version for compilation

Please correct if we are doing anything wrong since we are new to this GUI

VincentWei, 2022-12-15:
If you are doing cross building, there is no need to compile `cell-phone-ux-demo`.

gowrirengu, 2022-12-15:
Hi, 
Thank you for reply

As per the document we have taken the arm cross build directory as a reference for our mips cross compilation. For cross compilation its seems like we have to configure as per the platform.

1. For the cross compiling process we need to mention the directory in the modify-this-config.sh. but can you please clarify in TINA_DIR whether we need to mention the kernel source?

2. And also in the build-all.sh we understood we have to mention the mips toolchain under TOOLCHAIN_DIR, but we are not sure what we need to mention for HEADER_DIR

3. Do we need to do any additional step to configure or cross compile the MINIGUI.

4. As per our requirement we are going to use small display which will be used for displaying the live video. Is there any demo app available currently?

VincentWei, 2022-12-15:
As a start, you can just build MiniGUI core library (this repo) and a "hello, world" program. You can also refer to the following document for more information.

https://github.com/VincentWei/minigui-docs/blob/master/user-manual/README.md

> As per our requirement we are going to use small display which will be used for displaying the live video. Is there any demo app available currently?

No such demo is available.

VincentWei, 2022-12-15:
As a start, you can just build MiniGUI core library (this repo) and a "hello, world" program. You can also refer to the following document for more information:

https://github.com/VincentWei/minigui-docs/blob/master/user-manual/README.md

> As per our requirement we are going to use small display which will be used for displaying the live video. Is there any demo app available currently?

No such demo is available.