VincentWei/MiniGUI #119: Arch Linux 下编译 MiniGUI 失败

taotieren, 2023-06-18:
`minigui-git` PKGBUILD
```bash
# Contributor: taotieren <admin@taotieren.com>

pkgname=minigui-git
pkgver=5.0.11.r12.g0d1f1f02
pkgrel=1
pkgdesc="A modern and proven cross-platform window system for embedded systems or smart IoT devices."
arch=('any')
url="https://github.com/VincentWei/MiniGUI"
license=('GPLv3')
provides=(${pkgname%-git})
conflicts=(${pkgname%-git})
#replaces=(${pkgname})
depends=(
# chipmunk
        freetype2
        harfbuzz
        libdrm
        libevdev
        libjpeg-turbo
        libinput
        libpng
        kmod
        mtdev
        util-linux
        sqlite
        systemd
        pkgconf
        zlib)
optdepends=()
makedepends=(git
            cmake
            ninja)
backup=()
options=('!strip')
#install=${pkgname}.install
source=("${pkgname}::git+${url}.git")
sha256sums=('SKIP')

pkgver() {
    cd "${srcdir}/${pkgname}"
    git describe --long --tags | sed 's/ver-//g;s/\([^-]*-g\)/r\1/;s/-/./g'
}

build() {
    cd "${srcdir}/${pkgname}"

# see：https://wiki.archlinux.org/title/CMake_package_guidelines
#     cmake -DCMAKE_BUILD_TYPE=Release \
    cmake -DCMAKE_BUILD_TYPE=None \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=lib \
        -DCMAKE_INSTALL_LIBEXECDIR=lib \
        -B build \
        -G Ninja

    ninja -C build
}

package() {
    DESTDIR="${pkgdir}" ninja -C "${srcdir}"/${pkgname}/build install
}

```
 build log 

```bash
minigui-git on  master [?] took 14s 
❯ extra-x86_64-build
[... 480 lines cut ...]
!! WARNING
!!
!!
!! Product ID is *DISABLE*
!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

-- Configuring done (5.5s)
CMake Error at src/CMakeLists.txt:106 (add_library):
  Cannot find source file:

    misc/listenfd.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at src/CMakeLists.txt:106 (add_library):
  No SOURCES given to target: _ths

CMake Generate step failed.  Build files cannot be regenerated correctly.
==> ERROR: A failure occurred in build().
    Aborting...
==> ERROR: Build failed, check /var/lib/archbuild/extra-x86_64/taotieren/build

```

taotieren, 2023-06-28:
```bash
❯ makepkg -sf
==> 正在创建软件包：minigui-git 5.0.11.r12.g0d1f1f02-1 (2023年06月28日 星期三 21时04分33秒)
==> 正在检查运行时依赖关系...
==> 正在检查编译时依赖关系==> 获取源代码...
  -> 正在升级 minigui-git git 仓库...
==> 正在验证 source 文件，使用sha256sums...
    minigui-git ... 已跳过==> 正在释放源码...
  -> 正在建立 minigui-git git 仓库的拷贝...
正克隆到 'minigui-git'...
完成。==> 正在开始 pkgver()...
==> 正在开始 build()...
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 13.1.1
-- The CXX compiler identification is GNU 13.1.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Find PTHREAD include in /usr/include
-- Find PTHREAD library in /usr/lib/libpthread.a
-- Performing Test _compile_ret
-- Performing Test _compile_ret - Success
-- Looking for include file alloca.h
-- Looking for include file alloca.h - found
-- Looking for include file dirent.h
-- Looking for include file dirent.h - found
-- Looking for include file dlfcn.h
-- Looking for include file dlfcn.h - found
-- Looking for include file inttypes.h
-- Looking for include file inttypes.h - found
-- Looking for include file locale.h
-- Looking for include file locale.h - found
-- Looking for include file math.h
-- Looking for include file math.h - found
-- Looking for include file memory.h
-- Looking for include file memory.h - found
-- Looking for include file ndir.h
-- Looking for include file ndir.h - not found
-- Looking for include file stdint.h
-- Looking for include file stdint.h - found
-- Looking for include file stdlib.h
-- Looking for include file stdlib.h - found
-- Looking for include file strings.h
-- Looking for include file strings.h - found
-- Looking for include file string.h
-- Looking for include file string.h - found
-- Looking for include file sys/dir.h
-- Looking for include file sys/dir.h - found
-- Looking for include file sys/ndir.h
-- Looking for include file sys/ndir.h - not found
-- Looking for include file sys/stat.h
-- Looking for include file sys/stat.h - found
-- Looking for include file sys/time.h
-- Looking for include file sys/time.h - found
-- Looking for include file sys/types.h
-- Looking for include file sys/types.h - found
-- Looking for include file termio.h
[... 151 lines cut ...]
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!
!! WARNING
!!
!!
!! Product ID is *DISABLE*
!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

-- Configuring done (3.9s)
CMake Error at src/CMakeLists.txt:106 (add_library):
  Cannot find source file:

    misc/listenfd.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at src/CMakeLists.txt:106 (add_library):
  No SOURCES given to target: _ths

CMake Generate step failed.  Build files cannot be regenerated correctly.
==> 错误： 在 build() 中发生一个错误。    正在放弃...

```

VincentWei, 2023-06-29:
哦，您在用 cmake 配置和编译啊？cmake 不维护了，仍然在使用 autotools。所以您要用 autotools 那套命令来配置和编译。

taotieren, 2023-06-30:
用 `automake` 方式重写 PKGBUILD
```bash
# Maintainer:  taotieren <admin@taotieren.com>

pkgbase=minigui-git
pkgname=(minigui-{,{mgutils,mgplus,mgeff,mgncs,mgncs4touch,mg-tools,mg-tests,mg-samples,mg-demos,cell-phone-ux-demo,res,docs}-}git)
pkgver=5.0.11.r12.g0d1f1f02
pkgrel=1
arch=('any')
_url="https://github.com/VincentWei"
url="${_url}/MiniGUI"
license=('GPLv3')
provides=(${pkgname%-git})
conflicts=(${pkgname%-git})
# replaces=(${pkgname})
depends=(
        chipmunk
        freetype2
        harfbuzz
        libdrm
        libevdev
        libjpeg-turbo
        libinput
        libpng
        kmod
        mtdev
        util-linux
        sqlite
        systemd
        pkgconf
        zlib)
optdepends=()
makedepends=(git
            automake)
backup=()
options=('!strip')
install=
source=("${pkgbase}::git+${url}.git"
        "minigui-mgutils::git+${_url}/mgutils.git"
        "minigui-mgplus::git+${_url}/mgplus.git"
        "minigui-mgeff::git+${_url}/mgeff.git"
        "minigui-mgncs::git+${_url}/mgncs.git"
        "minigui-mgncs4touch::git+${_url}/mgncs4touch.git"
        "minigui-mg-tools::git+${_url}/mg-tools.git"
        "minigui-mg-tests::git+${_url}/mg-tests.git"
        "minigui-mg-samples::git+${_url}/mg-samples.git"
        "minigui-mg-demos::git+${_url}/mg-demos.git"
        "minigui-cell-phone-ux-demo::git+${_url}/cell-phone-ux-demo.git"
        "minigui-res::git+${_url}/minigui-res.git"
        "minigui-docs::git+${_url}/minigui-docs.git")

sha256sums=('SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP')

pkgver() {
    cd "${srcdir}/${pkgbase}"
    git describe --long --tags | sed 's/ver-//g;s/\([^-]*-g\)/r\1/;s/-/./g'
}

[... 188 lines cut ...]
    pkgdesc="Documents for MiniGUI Core and MiniGUI Components."

    install -dm0755 ${pkgdir}/usr/share/doc
    cp -rv "${srcdir}"/${pkgname%-git}  "${pkgdir}"/usr/share/doc/
}

package_minigui-git() {
    pkgdesc="A modern and proven cross-platform window system for embedded systems or smart IoT devices."
    depends=(minigui-{mgutils,mgplus,mgeff,mgncs,mgncs4touch,mg-tools,mg-tests,mg-samples,mg-demos,cell-phone-ux-demo,res,docs})

    source "${srcdir}/myconfig.sh"
    cd "${srcdir}/${pkgbase}"
    ./autogen.sh
    ./configure --prefix=/usr \
                --disable-static $GOPTS

    make -j$NR_JOBS
    make DESTDIR="$pkgdir" install
}

```

编译时还是报错

[minigui-build-log.md](https://github.com/VincentWei/MiniGUI/files/11918804/minigui-build-log.md)

VincentWei, 2023-07-03:
可以参考下 build-minigui-5.0 中的配置和构建脚本。

taotieren, 2023-07-06:
更新 PKGBUILD 后
```bash
# Maintainer:  taotieren <admin@taotieren.com>

pkgbase=minigui-git
pkgname=(minigui{,-{mgutils,mgplus,mgeff,mgncs,chipmunk,mgncs4touch,mg-tools,mg-tests,mg-samples,mg-demos,cell-phone-ux-demo,res,docs,ministudo}}-git)
pkgver=5.0.11.r12.g0d1f1f02
pkgrel=1
arch=('x86_64')
_url="https://github.com/VincentWei"
url="${_url}/MiniGUI"
license=('GPLv3')
depends=(
        freetype2
        gtk2
        harfbuzz
        libdrm
        libevdev
        libjpeg-turbo
        libtool
        libinput
        libwebp
        libxml2
        libpng
        kmod
        mtdev
        openssl
        util-linux
        sqlite
        systemd
        xorg-server-xvfb
        zlib)
optdepends=()
makedepends=(git
            autoconf
            automake
            cmake
            make
            pkgconf)
backup=()
options=('!strip')
install=
source=("${pkgbase}::git+${url}.git"
        "minigui-mgutils::git+${_url}/mgutils.git"
        "minigui-mgplus::git+${_url}/mgplus.git"
        "minigui-mgeff::git+${_url}/mgeff.git"
        "minigui-mgncs::git+${_url}/mgncs.git"
        "minigui-mgncs4touch::git+${_url}/mgncs4touch.git"
        "minigui-mg-tools::git+${_url}/mg-tools.git"
        "minigui-mg-tests::git+${_url}/mg-tests.git"
        "minigui-mg-samples::git+${_url}/mg-samples.git"
        "minigui-mg-demos::git+${_url}/mg-demos.git"
        "minigui-chipmunk::git+${_url}/chipmunk.git"
        "minigui-cell-phone-ux-demo::git+${_url}/cell-phone-ux-demo.git"
        "minigui-res::git+${_url}/minigui-res.git"
        "minigui-docs::git+${_url}/minigui-docs.git"
        "minigui-ministudio::git+https://github.com/FMSoftCN/ministudo-guibuilder.git")

sha256sums=('SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
            'SKIP'
[... 540 lines cut ...]
  290 |     return(fdata);
      |           ^
bdf2bin.c:108:33: warning: unused variable ‘code’ [-Wunused-variable]
  108 |     int num, width, high, i, j, code = 0, pitch;
      |                                 ^~~~
bdf2bin.c:107:16: warning: unused variable ‘ch2’ [-Wunused-variable]
  107 |     u_char ch, ch2;
      |                ^~~
bdf2bin.c:107:12: warning: unused variable ‘ch’ [-Wunused-variable]
  107 |     u_char ch, ch2;
      |            ^~
bdf2bin.c:231:17: warning: ignoring return value of ‘fgets’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
  231 |                 fgets(line, 256, fp);
      |                 ^~~~~~~~~~~~~~~~~~~~
mv -f .deps/font.Tpo .deps/font.Po
cc1: some warnings being treated as errors
make[1]: *** [Makefile:385: bin2c.o] Error 1
make[1]: *** Waiting for unfinished jobs....
mv -f .deps/bdf2bin.Tpo .deps/bdf2bin.Po
make[1]: Leaving directory '/home/taotieren/git_clone/aur/minigui-git/src/minigui-mg-tools/tools'
make: *** [Makefile:364: all-recursive] Error 1
==> ERROR: A failure occurred in package_minigui-mg-tools-git().
    Aborting...

```