VincentWei/MiniGUI #120: 关于韩语异常显示的问题

htk719809837, 2023-06-30:
你好，在使用中发现韩语存在两个问题，比较紧急，希望能够帮忙尽快修复
1、Minigui 3.0.12 使用韩语：static控件，设置部分文本时（多行），行尾最后一个字会出现乱码现象，下一行的第一个字也会出现乱码。
2、Minigui 3.0.12 / 5.0.11 使用韩语发现：韩文尾部混合英文字母的时候（单行），使用drawtext进行绘制文本时，尾部字母没有正常显示出来。将该尾部字母移动到行首时，又可以正常显示。

VincentWei, 2023-07-03:
收到。请提供可重现问题的源代码。

htk719809837, 2023-07-04:
#include <stdio.h>
#include <minigui/common.h>
#include <minigui/minigui.h>
#include <minigui/gdi.h>
#include <minigui/window.h>
#include <minigui/control.h>
#include <time.h>
#include <sys/time.h>

#define MSG_TIMER_START (0x900)

static PLOGFONT caption_font;

static LRESULT HelloWinProc(HWND hWnd, UINT message, WPARAM wParam, LPARAM lParam)
{
    static HWND button;
    HWND hWnd1;
    switch (message) {
        case MSG_CREATE:
	hWnd1 = CreateWindowEx(CTRL_STATIC,
		"비밀번호 재설정 기능이 잠겼습니다.....30A1분 후 시도하십시오.",
		WS_VISIBLE | SS_CENTER,
		WS_EX_TRANSPARENT,
		111,
		65, 
		50,
		355,
		64,
		hWnd,0);
   	caption_font= CreateLogFont("ttf", "vgaoem", "UTF-8", 
                     FONT_WEIGHT_REGULAR,
                    FONT_SLANT_ROMAN, FONT_SETWIDTH_NORMAL,
                    FONT_SPACING_CHARCELL, FONT_UNDERLINE_NONE,
                    FONT_STRUCKOUT_NONE, 18, 0);
        SetWindowFont(hWnd1, caption_font);
	SetWindowElementAttr(hWnd1, WE_FGC_WINDOW, 0x00ff00);
            break;

        case MSG_CLOSE:
            DestroyMainWindow (hWnd);
            PostQuitMessage (hWnd);
            break;
    }

    return DefaultMainWinProc(hWnd, message, wParam, lParam);
}

int MiniGUIMain (int argc, const char* argv[])
{
    MSG Msg;
    HWND hMainWnd;
    MAINWINCREATE CreateInfo;

    JoinLayer(NAME_DEF_LAYER , "timer" , 0 , 0);

    CreateInfo.dwStyle = WS_VISIBLE | WS_BORDER | WS_CAPTION;
    CreateInfo.dwExStyle = WS_EX_NONE;
    CreateInfo.spCaption = "Hello, world";
    CreateInfo.hMenu = 0;
    CreateInfo.hCursor = GetSystemCursor(0);
    CreateInfo.hIcon = 0;
    CreateInfo.MainWindowProc = HelloWinProc;
    CreateInfo.lx = 0;
    CreateInfo.ty = 0;
    CreateInfo.rx = 600;
    CreateInfo.by = 600;
    CreateInfo.iBkColor = COLOR_lightwhite;
    CreateInfo.dwAddData = 0;
    CreateInfo.hHosting = HWND_DESKTOP;

    hMainWnd = CreateMainWindow (&CreateInfo);

    if (hMainWnd == HWND_INVALID)
        return -1;

    ShowWindow(hMainWnd, SW_SHOWNORMAL);

    while (GetMessage(&Msg, hMainWnd)) {
        TranslateMessage(&Msg);
        DispatchMessage(&Msg);
    }

    MainWindowThreadCleanup (hMainWnd);

    return 0;
}

htk719809837, 2023-07-04:
以上是问题一的复现代码，使用不等宽字体，在32位的环境下，5.0.11版本，（和#116 问题单的环境一样），若无法复现，可稍微调整韩语中间字母、数字和标点的数量，以及控件的宽度，即可复现

VincentWei, 2023-07-04:
收到。我们尽快处理。

VincentWei, 2023-07-05:
已处理，原因是文本中包含有 0x001E 控制字符，这类字符在渲染时将被忽略。rel-3-0 分支上也做了同步处理。

closed 2025-01-22 by VincentWei