VincentWei/MiniGUI #123: 控件的visible属性修改

AfterWindStop, 2023-09-01:
背景说明：创建了一个弹窗，有消息的时候这个弹窗在屏幕正中间弹出显示， 然后有一个“知道了”的按钮点击后该弹窗隐藏，弹窗的显示和隐藏是通过ShowWindow--SW_HIDE/SW_SHOWNORMAL来实现， 但是执行SW_HIDE后会有一个淡淡弹窗残留
[image]

弹窗的属性如下图：
[image]

使用版本：minigui3.2

是否有方法或者增加一个这样的方法来直接修改控件的visible属性来控制控件的显示和隐藏

此问题较紧急，希望能尽快提供一个解决方案，感谢！

AfterWindStop, 2023-09-01:
@VincentWei

VincentWei, 2023-09-01:
用 miniStudio 开发的？着急的话，可以找飞漫的付费技术支持：support@minigui.com。

AfterWindStop, 2023-09-01:
> 用 miniStudio 开发的？着急的话，可以找飞漫的付费技术支持：[support@minigui.com](mailto:support@minigui.com)。

嗯嗯， 是的， 用miniStudio来开发的， 目前刚接触ministudio不太熟，是否存在一个方法来直接来设置控件的visible属性呢或者说通过这种方式来实现控件的隐藏显示是否可行呢， 感谢

AfterWindStop, 2023-09-07:
> > 用 miniStudio 开发的？着急的话，可以找飞漫的付费技术支持：[support@minigui.com](mailto:support@minigui.com)。
> 
> 嗯嗯， 是的， 用miniStudio来开发的， 目前刚接触ministudio不太熟，是否存在一个方法来直接来设置控件的visible属性呢或者说通过这种方式来实现控件的隐藏显示是否可行呢， 感谢

@VincentWei 可以修改控件visible属性吗