VincentWei/MiniGUI #130: 俄语和塞尔维亚语显示问题

bjNong, 2024-02-01:
魏老师，你好！

我在使用5.0.13版本的代码发现俄语和塞尔维亚语drawtext等文字显示接口或单行编辑框（CTRL_SLEDIT）没有按照实际情况去居中显示。不知道具体是什么情况。

测试发现，在较大分辨率的显示屏上是比较友好的情况（720x1280），在320x240或者240x320上面，会有稍微偏移；在272x480分辨率上就偏移的很多了，在单行编辑框里面选择内容看起来和乱码一样。
[image]
[image]
[image]

使用的字体是
name1=ttf-arab-rrncnn-0-0-ISO8859-6,UTF-8,ISO8859-8,CP1256
fontfile1=/mnt/mtdblock/miniguires/res/font/timesbd.ttf

 CreateLogFont ("ttf", "arab", "UTF-8",FONT_WEIGHT_BOOK, FONT_SLANT_ROMAN,
						FONT_FLIP_NIL, FONT_OTHER_NIL, FONT_UNDERLINE_NONE, FONT_STRUCKOUT_NONE, 16, 0);

VincentWei, 2024-02-01:
这类问题比较难定位问题所在。贵司是我们的商业许可客户吗？如果不是，请致信 sales@minigui.com。

bjNong, 2024-02-01:
找到问题了，业务不小心操作了font->size大小导致